jlab-sensing/ENTS-node-firmware #5: Implement SDI-12 protocol

jmadden173, 2023-11-10:
SDI-12 protocol will read measurements from the TEROS-12 sensor

jmadden173, 2024-01-08:
Resources

TEROS-12 Sensor:
https://www.metergroup.com/en/meter-environment/products/teros-12-soil-moisture-sensor

Integration which describes SDI-12 commands for the TEROS-12 
https://publications.metergroup.com/Integrator%20Guide/18224%20TEROS%2011-12%20Integrator%20Guide.pdf

SDI-12 Specs:
https://www.sdi-12.org/current_specification/SDI-12%20Specification%201.4%20February%2020%202023.pdf

jmadden173, 2024-01-23:
Pins defined, but still not working. Next steps are connecting to an oscope. After work with @jmadden173 to look deeper.

Due date: 2024-02-02

cjosephson, 2024-02-01:
@stevegtaylor please don't create new issues in the future, simply reassign the issue to yourself and change the labels as necessary. this aligns better with industry standards, and is important for tracking the progress across the entire duration of the work cycle.

closed 2024-02-01 by cjosephson