jlab-sensing/ENTS-node-firmware #37: Files not being generated when moving LoRa firmware from Linux to Windows

stevegtaylor, 2024-02-12:
` #include "subghz_phy_version.h"
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\stm32\src\lora_app.o] Error 1`

Dependency graph on @stevegtaylor's windows machine.

jmadden173, 2024-02-12:
This is the gcc commands for `lora_app.o`
```
arm-none-eabi-gcc -o .pio/build/stm32/src/lora_app.o -c -save-temps=obj -mthumb -mcpu=cortex-m4 -Os -ffunction-sections -fdata-sections -Wall -nostdlib -DPLATFORMIO=60107 -DSTM32WLxx -DSTM32WLE5xx -DDMA_CCR_SECM -DDMA_CCR_PRIV -DSENSOR_ENABLED=0 -DUSE_BSP_DRIVER -DUSE_HAL_DRIVER -DF_CPU=48000000L -IInc -ISrc -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Utilities/lpm/tiny_lpm -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/SubGHz_Phy -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/SubGHz_Phy/stm32_radio_driver -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/Conf -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/Crypto -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/LmHandler -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/LmHandler/Packages -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/Mac -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/Mac/Region -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Middlewares/Third_Party/LoRaWAN/Utilities -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Utilities/timer -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Utilities/sequencer -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Utilities/trace/adv_trace -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Utilities/misc -I/home/jtmadden/repos/jlab/soil-power-sensor-firmware/proto/c/include -I/home/jtmadden/repos/jlab/soil-power-sensor-firmware/proto/c/src -ISrc -IInc -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Drivers/CMSIS/Include -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Drivers/CMSIS/Device/ST/STM32WLxx/Include -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Drivers/STM32WLxx_HAL_Driver/Inc -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Drivers/STM32WLxx_HAL_Driver/Src -I/home/jtmadden/.platformio/packages/framework-stm32cubewl/Drivers/CMSIS/DSP/Include Src/lora_app.c
```

stevegtaylor, 2024-02-13:
`arm-none-eabi-gcc -o .pio\build\stm32\src\lora_app.o -c -save-temps=obj -w -mthumb -mcpu=cortex-m4 -Os -ffunction-sections -fdata-sections -Wall -nostdlib -DPLATFORMIO=60113 -DSTM32WLxx -DSTM32WLE5xx -DDMA_CCR_SECM -DDMA_CCR_PRIV -DSENSOR_ENABLED=0 -DUSE_BSP_DRIVER -DUSE_HAL_DRIVER -DF_CPU=48000000L -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Utilities\lpm\tiny_lpm -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlUtilitieslpmtiny_lpm -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\SubGHz_Phy -ISrc -IInc -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlMiddlewaresThird_PartySubGHz_Phy -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlMiddlewaresThird_PartySubGHz_Phystm32_radio_driver -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN\Conf -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN\Crypto -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN\LmHandler -IC:\Users\Steve\.platformio\platforms\ststm32\builder\LmHandlerPackages -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN\Mac -IC:\Users\Steve\.platformio\platforms\ststm32\builder\MacRegion -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Middlewares\Third_Party\LoRaWAN\Utilities -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Utilities\timer -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlUtilitiestimer -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Utilities\sequencer -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlUtilitiessequencer -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Utilities\trace\adv_trace -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlUtilitiestraceadv_trace -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Utilities\misc -IC:\Users\Steve\.platformio\platforms\ststm32\builder\UsersSteve.platformiopackagesframework-stm32cubewlUtilitiesmisc -ISrc -IInc -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Drivers\CMSIS\Include -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Drivers\CMSIS\Device\ST\STM32WLxx\Include -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Drivers\STM32WLxx_HAL_Driver\Inc -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Drivers\STM32WLxx_HAL_Driver\Src -IC:\Users\Steve\.platformio\packages\framework-stm32cubewl\Drivers\CMSIS\DSP\Include Src\lora_app.c`

stevegtaylor, 2024-02-14:
```
if os.path.isdir(subghz_phy_dir):
    build_custom_lib(
        subghz_phy_dir,
        {
            "name": "Middleware-SubGHz_Phy",
            "dependencies": [
                {"name": "Utilities-misc"},
                {"name": "Utilities-trace-adv_trace"},
                {"name": "Utilities-timer"}
            ],
            "build": {
                "srcDir": ".",
                "includeDir": ".",
                "srcFilter": ["+<*.c>", "+<stm32_radio_driver/*.c>"],
                "flags": [
                    "-I $PROJECT_SRC_DIR",
                    "-I $PROJECT_INCLUDE_DIR",
                    f'-I "{subghz_phy_dir}"',
                    f'-I "{stm32_radio_driver_dir}"'
                    ],
                "libLDFMode": "deep",
            },
        },
    )
```
The issue turned out to be quoting file paths. Windows is less tolerant of character changes. There was something windows didn't like about how these were formatted.

changing from this
`f"-I {subghz_phy_dir}",
f"-I {stm32_radio_driver_dir}"`

to this 

`f'-I "{subghz_phy_dir}"',
f'-I "{stm32_radio_driver_dir}"'`

resolved the issue.

closed 2024-02-14 by stevegtaylor