jlab-sensing/ENTS-node-firmware #54: USB Interface for configuration and reading measurements

jmadden173, 2024-03-04:
I'm envisioning some CLI tool or python GUI interface that runs on a host machine that allows for easy setup of the device and reading measurements. Currently the only way to make changes is to flash a new version of the firmware. This is not ideal if someone doesn't have a background in EE. We already have protobuf support so we should be able to use that as a communication protocol between the device and laptop through UART.

The following issues should be complete before: #23

Kshitijpatil16, 2024-03-05:
Considering the use of a CLI tool and Python GUI interface to enhance user-friendliness, especially for non-EE collaborators, I have understood that the process of creating such a system should be
- Understanding and creating a protobuf file, compiling it using the protobuf compiler
- Integrating protobuf file into the device firmware
- Developing a Python code to establish a UART communication with the board and using a "pyserial" library to send and receive protobuf messages
- Develop a Python GUI and user commands in code to configure settings and receive measurements
- Generate protobuf messages of these commands and send them to the board over UART

Is this how this issue needs to be resolved?

jmadden173, 2024-05-14:
Here is a current list of required fields:

- Logger ID (`uint`)
- Cell ID (`uint`)
- WiFi SSID (`str`)
- WiFi Password (`str`)
- API Endpoint URL (`str`)
- Calibration V Slope (`double`)
- Calibration V Offset (`int`)
- Calibration I Slope (`double`)
- Calibration I Offset (`int`)

jmadden173, 2024-09-20:
Update list of required fields

Upload Settings
- Logger ID (`uint`)
- Cell ID (`uint`)
- Upload method (either LoRa or WiFi)
- Upload interval (`uint`)

Measurement Settings
- Enabled sensors (voltage, current, teros12, teros21, bme280)
- Calibration V Slope (`double`)
- Calibration V Offset (`double`)
- Calibration I Slope (`double`)
- Calibration I Offset (`double`)

WiFi Settings
- WiFi SSID (`str`), max length 32
- WiFi Password (`str`), max length 64
- API Endpoint URL (`str`), max length 256
- API Endpoint Port (`uint32`)

closed 2024-11-14 by jmadden173