jlab-sensing/ENTS-node-firmware #111: Fix linting errors

jmadden173, 2024-07-03:
With more people working on the project its finally time to following the linting guidelines. They'll help keep the style consistent throughout the project and make it easier for new people to understand the repo. The google c++ style guide has a configuration for `cpplint` and is included in `requirements.txt`. I listed everyone with a user code library that I felt was most applicable to what they are working on right now. For the library you are assigned the errors that show up in the linter will need to be fixed included corresponding example and test files. Each library should have its own branch name prefixed with `fix-lint-` and you can list me as a reviewer.

## Steps to lint

Create venv if not already created
```
python -m venv .venv
pip install -r requirements.txt
```

For stm32
```
cd stm32
cpplint --recursive lib Src/examples test
```

For esp32
```
cd esp32
cpplint --recursive --root=. Inc Src lib test
```

## Assignments

`ads` - @ahmedfalah01 
`battery` - @2004Island 
`fram` - @stevegtaylor 
`phytos31` - @Kshitijpatil16 
`sdi12` - @gxjacobs 
`sensors` - @jmadden173

stevegtaylor, 2024-07-10:
I have encountered some issues with the linting implementation.
1. cpplint parses STM32 generated code, and throws formatting errors. I personally think we shouldn't modify the generated code because it's just going to get overwritten on the next generation anyways.
ex.
`test\test_fram\test_fram.c:202:  { should almost always be at the end of the previous line  [whitespace/braces] [4]
test\test_fram\test_fram.c:217:  { should almost always be at the end of the previous line  [whitespace/braces] [4]
test\test_fram\test_fram.c:222:  { should almost always be at the end of the previous line  [whitespace/braces] [4]
test\test_fram\test_fram.c:236:  { should almost always be at the end of the previous line  [whitespace/braces] [4]`
corresponds to 
'  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
  HAL_RCC_MCOConfig(RCC_MCO1, RCC_MCO1SOURCE_SYSCLK, RCC_MCODIV_1);'

2. `Include the directory when naming header files  [build/include_subdir] [4]` using this format (by that I mean the forward '/') breaks the build on Windows.

3. cpplint does not like the use of `const variables` in arrays. `test\test_fifo\test_fifo.c:35:  Do not use variable-length arrays.  Use an appropriately named ('k' followed by CamelCase) compile-time constant for the size.  [runtime/arrays] [1]` was thrown by `uint8_t data[fram_buffer_size + 1];` where `fram_buffer_size` is defined as `static const uint16_t fram_buffer_size = FRAM_BUFFER_END - FRAM_BUFFER_START;`. This could be resolved by using `#define constant`. Unfortunately, this is at odds with our previously defined style guide. We will have to decide which style to use going forward.

jmadden173, 2024-07-18:
@stevegtaylor 
1. Yeah I agree. You unfortunately can't hardcode the filepaths in the configuration so you have to specify them on the cli. I pasted quick reference commands in the issue that do the entire project and can be adjusted to peoples specific libraries.
2. This also doesn't make sense in the context of our project structure and was causing other errors. I've disabled the error as part of (FIXME).
3. I don't think that's what the error is saying. `fram_buffer_size` gets calculated at compile time and is constant. I think that its getting annoyed at the `+1` inside the braces. I believe declaring it as a variable on the line before would resolve it.

jmadden173, 2024-07-18:
For any errors related to copyright, lets use the following format at the bottom of the doxygen file header.

```
 * @copyright
 * 
 * MIT License
 * 
 * Copyright (c) 2024 jLab in Smart Sensing at UCSC
 * 
 * Permission is hereby granted, free of charge, to any person obtaining a copy
 * of this software and associated documentation files (the "Software"), to deal
 * in the Software without restriction, including without limitation the rights
 * to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 * copies of the Software, and to permit persons to whom the Software is
 * furnished to do so, subject to the following conditions:
 * 
 * The above copyright notice and this permission notice shall be included in all
 * copies or substantial portions of the Software.
 * 
 * THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 * IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 * FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 * AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 * LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 * OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 * SOFTWARE.
 ```

jmadden173, 2024-07-19:
@stevegtaylor I was wrong about the variable length arrays. It requires it to be prefixed by a *k* and camel cased.

jmadden173, 2024-07-19:
Also everyone for the copyright message, I removed the message since it wasn't detecting the lines within the doxygen comments. The message should still be included in every file.

PR #135 (2024-09-26): Fix linting errors
**Name/Affiliation/Title**
John, UCSC, Maintainer

**Purpose of the PR**
- Fixes all linting errors for C/C++ code in `stm32`, `esp32`, `proto/c` projects
- Formats files with `clang-format` for code consistency
- Sets up vscode tasks for linting and formatting

**Development Environment**
`OS: Linux spruce 6.8.2-arch2-1 #1 SMP PREEMPT_DYNAMIC Thu, 28 Mar 2024 17:06:35 +0000 x86_64 GNU/Linux`
Hardware version: 2.2.1
Software version: branch
Platformio Version: 6.1.15

**Test Procedure**
Run through task list

**Additional Context**
N/A

**Task List**

- [x] Update `CHANGELOG [... cut]
closed 2024-09-26 by jmadden173