jlab-sensing/ENTS-node-firmware #156: Switch MCU to STM32WLE5JC

jmadden173, 2024-11-18:
So we have the wrong MCU defined for the board. Should be `STM32WLE5JC` instead of `STM32WL55JC`. The `WL55` is dual core while `WLE5` is single core so both use the same Arm M4 core, so we shouldn't have to modify too much. Unsure if we need to regenerate code from CubeMX for updated clock configuration.

Could also be related to issues with UART and peripherials due to bad clock configuration. See #148, #149 for some possible related issues.

I'm amazed that the code works with the wrong MCU defined.

jlin143, 2024-12-03:
I've been exploring a bit, and here's some options that I think could work:

1. (I'll be pursuing this option) Follow the migration guide to change MCUs using CubeMX and CubeIDE: https://community.st.com/t5/stm32-mcus/mcu1-to-mcu2-migration-option-within-stm32cubeide/ta-p/49650
3. Copy and modify the Wio-E5 default project to match the requirements defined in the existing `stm32.ioc` file.
4. Use `stm32pio` to regenerate the files from scratch.

jlin143, 2024-12-10:
As of 7c68857, the MCU has been updated to the STM32WLE5JCI6 and successfully builds `stm32`, `example_battery`, `example_adc`, `example_phytos`, `calibrate_adc`, and `example_sdi12`. When building for `test`, the linker faces errors. Even manually adding the linker file `board_build.ldscript = STM32WLE5JCIX_FLASH.ld` (both with and without READONLY sections), the linker still fails to build. It often has undefined references to main and undefined references to Error_Handler (defined in main).

jmadden173, 2025-11-04:
When I redid the clock clock configuration in #268 / #252 I updated the MCU.

closed 2025-11-04 by jmadden173