jlab-sensing/ENTS-node-firmware #183: Packet compression

jmadden173, 2025-02-28:
From the linked paper, there could be savings to be had by implementing compression on-top of protobuf encoding and aggregation (#173). For the sysdesign paper we can implement compression on the packets and benchmark the latency and power consumption on our embedded device. The issue should include some interface to switch between algos and benchmark latency and power consumption.

https://ieeexplore.ieee.org/document/10330191

stevegtaylor, 2025-03-31:
After reading the paper and some background, I don't think implementing Hoffman is going to be feasible for a few reasons
1. Quote from the paper "_It’s important to highlight that, in this context, the comparison involving the Huffman algorithm excludes the use of serialization. This decision stems from the fact that after data serialization into binary format, applying the Huffman algorithm with an extended dictionary becomes challenging_" the authors did not implement Huffman on ProtoBuff serialized data, this is not only perhaps an unfair comparison but not realistic to our application
2. Huffman (to my understanding) works by generating a binary tree based on character frequency. This is possible in IoT **if you assume message structure remains constant.** I'm not sure we can make this assumption, well this might be true in an individual deployment it would require generating new binary trees every time a sensor is added, in addition it assumes that the decoder knows the binary tree. Meaning it would most likely have to be re-generated before each deployment.

Let me know what you think of my reasoning @jmadden173

jmadden173, 2025-04-01:
@stevegtaylor Yeah I see what you mean. My intention was to encode the table within the packet not requiring the message structure is constant.. As you said tho on serialized data there there would be a binary tree with possibly 255 entries which I think would be larger than our serialized data packet. 

I also looked into AC and ANS as they are both entropy based encoding. (1) AC seems to be limited by the size of the data. For our small packet size it should actually work well since we don't need super high precision given that our max length is 255 bytes. (2) ANS may have a similar problem where the probability distribution needs to be known beforehand by both the encoder/decoder. Basically more generalization of AC. Definitely higher bar to implement still. But I still think both of these are promising.

jmadden173, 2025-04-01:
Also some other logistical things:
1. For each implementation can you create a PR so that it's easier to manage contribution? If I or anyone else has the time they can work on others that way.
2. I'll also need to know your implementation details for some of the compression implementation so that I can implement them on the backend for decoding.

stevegtaylor, 2025-04-01:
@jmadden173 we could do two tables, like of algos that take pre-configuration and algos that don't. But that doesn't get around the issue of applying character based frequency searching to serialized data

stevegtaylor, 2025-04-01:
@jmadden173 another thing we should talk about is how we're going to benchmark this.

- Are we going to generate large amounts of Protobuf serialized dummy messages and compare data compression rates? That's the most similar to what our example paper did
- Are we going to try and measure power consumption with each algo running dummy messages?

Did you have an evaluation strategy in mind?

stevegtaylor, 2025-04-09:
LZQ is not in example paper, and might not exist. Did you mean LZW?

closed 2026-01-05 by jmadden173